lancos/ponyprog #3: hardware support

johndoe71rus, 2017-10-14:
does ponyprog support ch341A or PL2303 cheap chips? 
thanks

lancos, 2017-10-15:
No. May be in the future when we'll add support for USB adapters.

johndoe71rus, 2017-10-15:
for ch341 i use https://github.com/setarcos/ch341prog 
maybe you can use it too

eduard-x, 2017-10-18:
I think, in case of USB to serial adapters (like PL2303) under Linux it's easy to add into the list, but i don't know, is it equally simple under Windows?
Under Linux is the adapter to find under /dev/ttyUSBx

johndoe71rus, 2017-10-18:
ch341 has serial and parallel port interface.
they both can be represented under /dev/ttyUSBx

https://www.onetransistor.eu/2017/08/ch341a-mini-programmer-schematic.html

eduard-x, 2017-10-19:
Yes, but all old hardware solutions are very simple, without interface to microcontroller. I think, it's better to implement it with libusb. I have bought this programmer on aliexpress, and can begin with tests 3-4 weeks later, after delivery. :)
Thanks for ch341prog source code link!

lancos, 2017-10-19:
Thank you for the links. CH341 (like other chips, for example FTDI) support SPI and I2CBus.
The problem is that ponyprog implements the SPI and I2CBus via bit-banging and add the USB
adapters need a restructure that was planned several weeks ago, but I need to find the time to do it. Now that ponyprog was ported to QT it is the first item in the TODO list.

eduard-x, 2017-11-29:
Hi!
I have now the CH341, can begin with implementation on Linux.
Best regards,
Eduard

johndoe71rus, 2018-02-11:
@eduard-x 
How is the work going?

eduard-x, 2018-02-28:
Hello johndoe71rus, 
i was busy with porting of Qucs to Qt5 and now for 1-2 month i'm writing my project for mk1 cnc.
Possible, in may-june i'll try to integrate the CH341 interface.

JucaTec, 2018-09-09:
Hi. Any news about USB? Thank you.

eduard-x, 2018-10-14:
Hi!
in next 2-3 weeks i'll try to compile it and to test the flashing. Sorry for delaying.

eduard-x, 2019-02-07:
Hello! From this weekend i will continue to work on this usb extension. Was very busy last months at work.

eduard-x, 2019-02-21:
Hi!
At this time was extended the connection selecting popup for ch341a connection types: i2c, spi and uart.
In PonyProg is enabled autodetection of USB devices: 1a86:5512 and 1a86:5523 
I'm trying to connect to ch341a class. Possible, i'll test it on this weekend or next week. All sources are in this branch: wip/eduard/libusb-patches

eduard-x, 2019-03-25:
Hi!
I need your help in debugging. At this moment i have implemented in "libusb-patches" branch the functions for bit-banging. All HW connections are like here: [https://i2.wp.com/scienceprog.com/wp-content/uploads/2007i/24cXX/i2cserial.gif](url) but without z-diodes and instead RS232 connectors i use this board [https://hackaday.com/2018/02/21/linux-adds-ch341-gpio/](url) in UART mode
Description for connectors from my file in branch: 
```
/**
 * ch341a modem connections:
 *
 * rs232 -> ch341  description
 *     1 -> 18     data carrier detect
 *     2 -> 6      Rx
 *     3 -> 5      Tx
 *     4 -> 20     Data Terminal Ready
 *     5              Ground
 *     6 -> 16     Data Set Ready
 *     7 -> 21     Request To Send
 *     8 -> 15     Clear To Send
 *     9 -> 17     Ring Indicator
 **/
```

And i'm getting from PonyProg followed debugging list (project compiled with enabled debug option):

```SIProgInterface::OpenUSB( 6790 21795 ) IN *** Inst= false
ch341::Open( 6790 21795 ) 
ch341::Configure()
Open device [1a86:5523].
Claim interface 0
Device reported its revision [4.03]
SIProgInterface::OpenUSB() =  0  OUT
e2App::OpenBus() ** OpenUSB =  0
e2App::OpenBus() ** SetPower
e2App::OpenBus() ** Reset
I2CBus::Reset() - IN
I2CBus::SetDelay() =  5
I2CBus::Read( 0 ,  0x7fffffffc127 ,  0 ) - IN
I2CBus::StartRead( 0 ,  0x7fffffffc127 ,  0 ) - IN
I2CBus::StartRead() =  1 , err_no =  0  - OUT
I2CBus::Read() =  1 , err_no =  0  - OUT
I2CBus::Reset() - OUT
e2AppWinInfo::Read() ** OpenBus =  0
E24xx::Probe( 1 ) - IN
I2CBus::Read( 160 ,  0x7fffffffc18b ,  1 ) - IN
I2CBus::StartRead( 160 ,  0x7fffffffc18b ,  1 ) - IN
I2CBus::Read() =  0 , err_no =  -13  - OUT
I2CBus::Read( 162 ,  0x7fffffffc18b ,  1 ) - IN
I2CBus::StartRead( 162 ,  0x7fffffffc18b ,  1 ) - IN
I2CBus::Read() =  0 , err_no =  -13  - OUT
I2CBus::Read( 164 ,  0x7fffffffc18b ,  1 ) - IN
I2CBus::StartRead( 164 ,  0x7fffffffc18b ,  1 ) - IN
I2CBus::Read() =  0 , err_no =  -13  - OUT
I2CBus::Read( 166 ,  0x7fffffffc18b ,  1 ) - IN
I2CBus::StartRead( 166 ,  0x7fffffffc18b ,  1 ) - IN
I2CBus::Read() =  0 , err_no =  -13  - OUT
I2CBus::Read( 168 ,  0x7fffffffc18b ,  1 ) - IN
I2CBus::StartRead( 168 ,  0x7fffffffc18b ,  1 ) - IN
I2CBus::Read() =  0 , err_no =  -13  - OUT
I2CBus::Read( 170 ,  0x7fffffffc18b ,  1 ) - IN
I2CBus::StartRead( 170 ,  0x7fffffffc18b ,  1 ) - IN
I2CBus::Read() =  0 , err_no =  -13  - OUT
I2CBus::Read( 172 ,  0x7fffffffc18b ,  1 ) - IN
I2CBus::StartRead( 172 ,  0x7fffffffc18b ,  1 ) - IN
I2CBus::Read() =  0 , err_no =  -13  - OUT
I2CBus::Read( 174 ,  0x7fffffffc18b ,  1 ) - IN
I2CBus::StartRead( 174 ,  0x7fffffffc18b ,  1 ) - IN
I2CBus::Read() =  0 , err_no =  -13  - OUT
E24xx::Probe() =  0  - OUT
e2App::CloseBus() iniBus= 0x7fffffffd758
I2CBus::Close() busI= 0x7fffffffd398
SIProgInterface::Close() IN *** Inst= true
SerialInterface::CloseSerial()
SIProgInterface::Close() OUT
e2AppWinInfo::Read() =  0  - OUT
CmdWindow->Read -- Error```

I'm trying to find the find the bug(s), but possible with more eyes we can find this faster :)

eduard-x, 2019-04-04:
Yes! I can read my 24x16 about the USB in UART mode! :)
For connection set in Interface PopUp window "Invert D-In", "Invert D-Out", "Invert SCKL" to checked.
The settings for Baudrate are now without any effect, this is not implemented.
The UART mode is direct communication without buffering. For this solution we don't need the original CH341 driver. Of cause, it is possible to extend later with I2C and SPI interfaces.
On weekend i will clean the source code in libusb-patches branch.

eduard-x, 2019-04-08:
Now i'm busy with additional Interfaces for ch341a: i2c, spi.

JucaTec, 2019-04-09:
Thanks for the work. Are you going to make software for Windows?

eduard-x, 2019-04-09:
Yes, after testing of USB integrated driver, libusb is an multiplatform library. But i don't know, what is with exec rights, i think, the program under Win must be start with admin rights. Under Linux is it easier to allow this with udev rules file for users.

johndoe71rus, 2019-06-21:
Try to build (lubuntu 18.04)
https://paste.ubuntu.com/p/KWVTnpzhZN/
i install libftdi-dev and libftdi1-dev

lancos, 2019-06-24:
You need to install libftdipp1-dev on 18.04

johndoe71rus, 2019-08-09:
i use ch341prog with ch341A only one time to read/write 25q64xxxxx 
this pcb https://images.ua.prom.st/879432686_w640_h640_usb-programmator-ch341a.jpg with 8 pin clip. 
Now, can i use branch "libusb-patches" try read/write 25q64xxxxx with ponyprog or not?

engmarcos26, 2020-01-07:
WHAT IS HW? COMPLET   TEXT, PLIS!

johndoe71rus, 2020-06-09:
I found a device with a weird usb serial bridge. defined as ID 1a86:7522 QinHeng Electronics. Google finds device ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter. I found the manufacturer page http://www.wch-ic.com/products/CH340.html similar to ch340k, 10 pin. For download, they suggest the file CH341SER_LINUX.ZIP but I can not build it because of an error. You can also find the file CH341PAR_LINUX.ZIP

christian-0s, 2020-06-28:
> I found a device with a weird usb serial bridge. defined as ID 1a86:7522 QinHeng Electronics. Google finds device ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter. I found the manufacturer [...]

You do not need another driver, just manually load module ch341 and then enter: 
sudo sh -c "echo 1a86 7522 > /sys/bus/usb-serial/drivers/ch341-uart/new_id"

johndoe71rus, 2020-06-30:
do you think so?
Linux driver CH341 don't know about ID 7522
https://github.com/torvalds/linux/blob/master/drivers/usb/serial/ch341.c#L78

christian-0s, 2020-06-30:
That's why you have to write it to new_id.
Tested on a LilyGO board based on ESP8266, which has a CH340K

johndoe71rus, 2020-06-30:
yep. Exactly this LILYGO board. Will try

GuilhermeMoreno, 2021-02-28:
@eduard-x Any news on the support of USB adapters for programming EEPROMs? Did you managed to get it working with CP2102?

mrv96, 2022-11-07:
I'm intrested too about CH341 support