lancos/ponyprog #6: Ponprog 3.0, CancelIOEx could not be located in the dynamic link library KERNEL32.dll

xiaolaba, 2018-05-01:
hi,
can you please help to look at this.
xp and win10, same response

lancos, 2018-05-02:
Hi,
PonyProg3 (Qt5.8) doesn't support WindowsXP. If you need WindowsXP you can build the old PonyProg2 legacy_v2 branch (https://github.com/lancos/ponyprog/tree/legacy_v2) without Qt.

closed 2018-05-02 by lancos

xiaolaba, 2018-05-03:
thanks you Sir.got it
best regards,xiao

      寄件人︰ Claudio Lanconelli <notifications@github.com>
 收件人︰ lancos/ponyprog <ponyprog@noreply.github.com> 
副本(CC)︰ xiaolaba <wayen_cwwong@yahoo.com>; Author <author@noreply.github.com>
 傳送日期︰ 2018年05月2日 (週三) 10:13 PM
 主題︰ Re: [lancos/ponyprog] Ponyprog 3.0, CancelIOEx could not be located in the dynamic link library KERNEL32.dll (#6)
   
Hi,
PonyProg3 (Qt5.8) doesn't support WindowsXP. If you need WindowsXP you can build the old PonyProg2 legacy_v2 branch (https://github.com/lancos/ponyprog/tree/legacy_v2) without Qt.—

eduard-x, 2018-05-03:
Hi xiaolaba,
You can build the project for Windows XP with Qt4.8. This is possible, but you must switch the CMakeLists for Qt4: OPTION (USE_QT5 "Using of Qt5 version for compiling" OFF)
Regards,
Eduard

xiaolaba, 2018-05-05:
hello Eduard
We had a shot with qt-5.7, used migwin-32 for compiling, it is ok with win10 & test, but not yet carry over to XP machine, however, one more question is about the dll dynamic loading but not single exe file. it is something that no easy for user unless copy those DLLs to target machine, looks like qt is no default switch for user to do such extends.
we will try qt4.8 as you have suggested.

best regards,
xiao