lancos/ponyprog #7: Move to qt5

johndoe71rus, 2018-12-18:
Kde team said
Currently Qt4 has been dead upstream and we are starting to have problems
maintaining it" 

Now qt4 is removed from different distributions. Offer to switch to qt5.
log from lubuntu 18.04
```
./create_deb_package.sh 
cpu cores: 4
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Deprecation Warning at CMakeLists.txt:5 (CMAKE_POLICY):
  The OLD behavior for policy CMP0003 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

CMake Deprecation Warning at CMakeLists.txt:6 (CMAKE_POLICY):
  The OLD behavior for policy CMP0015 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Qt version for compiling: 4
-- CMAKE_CXX_FLAGS -O3 -DNDEBUG -Wall
-- Version info: 3.0.0
-- Version build date: 18.12.2018
-- Compile without profiling information
CMake Error at /usr/share/cmake-3.10/Modules/FindQt4.cmake:1320 (message):
  Found unsuitable Qt version "5.9.5" from /usr/bin/qmake, this code requires
  Qt 4.x
Call Stack (most recent call first):
  CMakeLists.txt:143 (FIND_PACKAGE)

-- Configuring incomplete, errors occurred!
See also "/home/johndoe/build/ponyprog/ponyprog/build-deb/CMakeFiles/CMakeOutput.log".
make: *** No targets specified and no makefile found.  Stop.
strip: 'ponyprog': No such file
make: *** No rule to make target 'package-binary-deb'.  Stop.
ls: cannot access 'ponyprog*.deb': No such file or directory
mv: './build-deb/' and './build-deb' are the same file
dpkg-deb: error: --extract needs a target directory.
Perhaps you should be using dpkg --install ?

Type dpkg-deb --help for help about manipulating *.deb files;
Type dpkg --help for help about installing and deinstalling packages.
dpkg-deb: error: failed to read archive 'fix_up_deb/DEBIAN': No such file or directory
rm: missing operand
Try 'rm --help' for more information.
chmod: cannot access 'fix_up_deb/DEBIAN/md5sums': No such file or directory
sed: can't read fix_up_deb/DEBIAN/md5sums: No such file or directory
dpkg-deb: error: failed to open package info file 'fix_up_deb/DEBIAN/control' for reading: No such file or directory

lancos, 2019-02-06:
Try to change line 33 -DUSE_QT5=ON in create_deb_package.sh

lancos, 2019-02-06:
This is the log on ubuntu bionic 18.04 with -DUSE_QT5=ON:

./create_elf_binary.sh 
cpu cores: 4
-- The C compiler identification is GNU 7.3.0
-- The CXX compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Qt version for compiling: 5
-- CMAKE_CXX_FLAGS -O3 -DNDEBUG -Wall
-- Version info: 3.0.1
-- Version build date: 06.02.2019
-- Compile without profiling information
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/Qt5/Qt5Config.cmake:28 (find_package):
  Could not find a package configuration file provided by "Qt5Multimedia"
  with any of the following names:

    Qt5MultimediaConfig.cmake
    qt5multimedia-config.cmake

  Add the installation prefix of "Qt5Multimedia" to CMAKE_PREFIX_PATH or set
  "Qt5Multimedia_DIR" to a directory containing one of the above files.  If
  "Qt5Multimedia" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:151 (FIND_PACKAGE)

-- Configuring incomplete, errors occurred!
See also "/home/lancos/Projects/ponyprog/build/CMakeFiles/CMakeOutput.log".

Eduard, can you suggest how to fix cmake, please?

johndoe71rus, 2019-02-07:
you need https://packages.ubuntu.com/en/bionic/qtmultimedia5-dev

eduard-x, 2019-02-07:
Hi! In the CMakeLists.txt i can add the line with message about package name, when not found. But the package name possible depended from distribution.

lancos, 2019-02-07:
Yes, qtmuldimedia5-dev solved the issue, thank you. The name of the package doesn't help: libqt5multimedia5 vs qtmultimedia5-dev instead of libqt5multimedia5-dev

closed 2019-02-07 by lancos

anlyoung, 2021-09-12:
I have a similar problem while catkin_make a simulation package for a robot system (Baxter by Rethink robotics):
-- +++ processing catkin package: 'baxter_sim_io'
-- ==> add_subdirectory(baxter_simulator/baxter_sim_io)
CMake Error at /usr/share/cmake-3.10/Modules/FindQt4.cmake:1320 (message):
  Found unsuitable Qt version "5.9.5" from /usr/bin/qmake, this code requires
  Qt 4.x
Call Stack (most recent call first):
  baxter_simulator/baxter_sim_io/CMakeLists.txt:15 (find_package)

What/where is create_deb_package.sh file?

anlyoung, 2021-09-12:
I revised the CMakeLists.txt file:  from 'find_package(Qt4 REQUIRED COMPONENTS QtCore QtGui)' to 'find_package(Qt5 REQUIRED COMPONENTS QtCore QtGui)' and got the following error message:
------------------------------------------------
-- +++ processing catkin package: 'baxter_sim_io'
-- ==> add_subdirectory(baxter_simulator/baxter_sim_io)
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/Qt5/Qt5Config.cmake:28 (find_package):
  Could not find a package configuration file provided by "Qt5QtCore" with
  any of the following names:

    Qt5QtCoreConfig.cmake
    qt5qtcore-config.cmake

  Add the installation prefix of "Qt5QtCore" to CMAKE_PREFIX_PATH or set
  "Qt5QtCore_DIR" to a directory containing one of the above files.  If
  "Qt5QtCore" provides a separate development package or SDK, be sure it has
  been installed.
----------------------------------------------------------------

anlyoung, 2021-09-13:
Resolved.. by following general instructions above
https://stackoverflow.com/questions/50739201/update-cmakelists-qt4-to-qt5