lancos/ponyprog #9: compilation issues on various platforms

tijuca, 2019-02-06:
Hi,

after the ponyprog package was accepted into the unstable archive of Debian the build daemons have spotted some compilation issues. Mainly to a to generic usage of `#ifdef __linux__`.

https://buildd.debian.org/status/package.php?p=ponyprog

For the Buster release only the white filled rows are important because these are the release critical platforms. So far I can see now the inclusion of `sys/io.h` in the file `SrcPony/ponyioint.cpp` is problematic as this header is highly architecture depended. A possible short time solution would be to include the needed functions from this header into a own header or cpp file. If this isn't possible the build support should be decreased to platforms which are supported.

I'm happy to help out with testing the compilation on the currently not working platforms, but I don't have a big C++ and in detail no QT experience.

lancos, 2019-02-06:
Hi,
pick-up master branch, I should have fixed all issues you reported.

tijuca, 2019-02-06:
Reading especially your last commit you have decreased the architectures to i386 and amd64? Right? If yes than I don't need much and consuming testing, both platforms builds I can check locally. Unfortunately I can find some more information about this within the commit messages so sorry for maybe stupid asking.

lancos, 2019-02-06:
Hi, ponyprog can use multiple different interfaces to access eeprom/flash chips, if depends on the hardware where you run ponyprog. Old PC can access gpio I/O via sys/io.h (ioperm, inb, outb), it needs root interface, and don't work on arch like ARM, MIPS, etc. I could have removed completely this interface or just limit it to X86 PC.
However other interfaces (for example the one that use sysfs) work well on ARM like raspberry and embedded boards, I suppose the same is valid for MIPS, but I don't know.
So don't remove all the ARCHs, please, at least keep X86 and ARM (both 32bit and 64bit).
In the future (probably version 3.1.0) I plan to add also USB adapters so every ARCH will be supported.

tijuca, 2019-02-06:
O.K. so no Intel based platforms only, fine.
I will pick up some time on the weekend to do some test builds then on at least on the release critical platforms.

lancos, 2019-02-06:
thank you for your time.

tijuca, 2019-02-07:
Hmm, I'd like to keep this issue open until we have sorted out all things. Otherwise I need to create always a new issue for the various issues. I guess this would be unfortunate.

I started to go onto a mispel system (32bit Mips LE) with the current head in master. The build breaks in `SrcPony/portint.cpp`. This will happen on all Mips platforms I bet, namely mips, mipsel, mips64le.

```
[ 76%] Building CXX object CMakeFiles/ponyprog.dir/SrcPony/at250xx.cpp.o
/usr/bin/c++  -DQT_CORE_LIB -DQT_GUI_LIB -DQT_MULTIMEDIA_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_PRINTSUPPORT_LIB -DQT_PROJECT -DQT_WIDGETS_LIB -I/home/tijuca/ponyprog_3.0.1+20190206+ds -I/home/tijuca/ponyprog_3.0.1+20190206+ds/qhexedit2/src -I/home/tijuca/ponyprog_3.0.1+20190206+ds/SrcPony -I/home/tijuca/ponyprog_3.0.1+20190206+ds/debian/build -isystem /usr/include/mipsel-linux-gnu/qt5 -isystem /usr/include/mipsel-linux-gnu/qt5/QtWidgets -isystem /usr/include/mipsel-linux-gnu/qt5/QtGui -isystem /usr/include/mipsel-linux-gnu/qt5/QtCore -isystem /usr/lib/mipsel-linux-gnu/qt5/mkspecs/linux-g++ -isystem /usr/include/mipsel-linux-gnu/qt5/QtPrintSupport -isystem /usr/include/mipsel-linux-gnu/qt5/QtMultimedia -isystem /usr/include/mipsel-linux-gnu/qt5/QtNetwork  -g -Wall -std=c++11 -fno-exceptions -fno-rtti      -fPIC -std=gnu++11 -o CMakeFiles/ponyprog.dir/SrcPony/at250xx.cpp.o -c /home/tijuca/ponyprog_3.0.1+20190206+ds/SrcPony/at250xx.cpp
/home/tijuca/ponyprog_3.0.1+20190206+ds/SrcPony/portint.cpp: In member function 'virtual int PortInterface::InPort(int) const':
/home/tijuca/ponyprog_3.0.1+20190206+ds/SrcPony/portint.cpp:209:2: error: return-statement with no value, in function returning 'int' [-fpermissive]
  return inb(nport);
  ^~~~~~
/home/tijuca/ponyprog_3.0.1+20190206+ds/SrcPony/portint.cpp: In member function 'int PortInterface::OpenSerial(int)':
/home/tijuca/ponyprog_3.0.1+20190206+ds/SrcPony/portint.cpp:388:45: error: expected primary-expression before ';' token
     lcr_copy = inb(GetFirstPort() + lcrOfst);
                                             ^
/home/tijuca/ponyprog_3.0.1+20190206+ds/SrcPony/portint.cpp:390:45: error: expected primary-expression before ';' token
     ier_copy = inb(GetFirstPort() + ierOfst);
                                             ^
[ 77%] Building CXX object CMakeFiles/ponyprog.dir/SrcPony/at90sxx.cpp.o
/usr/bin/c++  -DQT_CORE_LIB -DQT_GUI_LIB -DQT_MULTIMEDIA_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_PRINTSUPPORT_LIB -DQT_PROJECT -DQT_WIDGETS_LIB -I/home/tijuca/ponyprog_3.0.1+20190206+ds -I/home/tijuca/ponyprog_3.0.1+20190206+ds/qhexedit2/src -I/home/tijuca/ponyprog_3.0.1+20190206+ds/SrcPony -I/home/tijuca/ponyprog_3.0.1+20190206+ds/debian/build -isystem /usr/include/mipsel-linux-gnu/qt5 -isystem /usr/include/mipsel-linux-gnu/qt5/QtWidgets -isystem /usr/include/mipsel-linux-gnu/qt5/QtGui -isystem /usr/include/mipsel-linux-gnu/qt5/QtCore -isystem /usr/lib/mipsel-linux-gnu/qt5/mkspecs/linux-g++ -isystem /usr/include/mipsel-linux-gnu/qt5/QtPrintSupport -isystem /usr/include/mipsel-linux-gnu/qt5/QtMultimedia -isystem /usr/include/mipsel-linux-gnu/qt5/QtNetwork  -g -Wall -std=c++11 -fno-exceptions -fno-rtti      -fPIC -std=gnu++11 -o CMakeFiles/ponyprog.dir/SrcPony/at90sxx.cpp.o -c /home/tijuca/ponyprog_3.0.1+20190206+ds/SrcPony/at90sxx.cpp
make[4]: *** [CMakeFiles/ponyprog.dir/build.make:974: CMakeFiles/ponyprog.dir/SrcPony/portint.cpp.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[4]: Leaving directory '/home/tijuca/ponyprog_3.0.1+20190206+ds/debian/build'
make[3]: *** [CMakeFiles/Makefile2:108: CMakeFiles/ponyprog.dir/all] Error 2
make[3]: Leaving directory '/home/tijuca/ponyprog_3.0.1+20190206+ds/debian/build'
make[2]: *** [Makefile:133: all] Error 2
make[2]: Leaving directory '/home/tijuca/ponyprog_3.0.1+20190206+ds/debian/build'
dh_auto_build: cd debian/build && make -j4 "INSTALL=install --strip-program=true" returned exit code 2
make[1]: *** [debian/rules:45: override_dh_auto_build] Error 2
make[1]: Leaving directory '/home/tijuca/ponyprog_3.0.1+20190206+ds'
make: *** [debian/rules:31: build] Error 2
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
```

I can keep the build open and do some kind of online try&error replacing if you can say me what I should change.

lancos, 2019-02-07:
Sorry, try this  patch

```
diff --git a/SrcPony/portint.cpp b/SrcPony/portint.cpp
index 3cbabe0..1533ae0 100644
--- a/SrcPony/portint.cpp
+++ b/SrcPony/portint.cpp
@@ -65,7 +65,7 @@ int PortInterface::IOperm(int a, int b, int c)
 }
 
 #define outb(x, p)
-#define inb(p)
+#define inb(p)	0
 #endif
 
 #endif
-- 
2.17.1

```
[0001-Fix-missing-ret-value.patch.txt](https://github.com/lancos/ponyprog/files/2842554/0001-Fix-missing-ret-value.patch.txt)

tijuca, 2019-02-07:
Looks good, the build is now successful on mispel. But I will recheck the other platforms too before saying something finally.

BTW: Another small thing I noticed in the builds is a bit of over linking.

`dpkg-shlibdeps: warning: package could avoid a useless dependency if debian/ponyprog/usr/bin/ponyprog was not linked against libQt5Network.so.5 (it uses none of the library's symbols)`

Maybe you can have a look at this too, seems like this library isn't really needed (anymore?).

lancos, 2019-02-07:
The Qt5Multimedia depends on Qt5Network (don't ask me why, it's a qt stuff), both qt5 cmake and qmake add linking to libQt5Network.so.5 if you use Qt5Multimedia package.

eduard-x, 2019-02-08:
Possible because of multimedia streaming support, or something like this

tijuca, 2019-02-08:
Never mind then, it's more a upstream issue in QT I guess. There are more worse upstream projects around which do much more over linking.

Back to topic, I tested the build together with the patch on other platforms which doesn't have succeeded a build ever, but it looks fine on all of these. I've seen a successful build on arm64, mips, mipsel, mips64el, ppc64el and s390x. So all RC platforms seems to build ponyprog.

So do you mind then to do a version 3.0.2 or want to this later. If yes I would wait until a new tagged version is available.

Thanks for tanking care on my report!

lancos, 2019-02-09:
Hi,
I just add 3.0.2 tag

tijuca, 2019-02-09:
Thanks!

Just uploaded the updated package a few minutes ago. I guess all RC platforms should now go into green after some days. We will see. :smile: 

Thanks for the quick updates!
I'll close this issue by this comment.

closed 2019-02-09 by tijuca