lancos/ponyprog #16: Not install in archlinux x86_64

darkprof83, 2020-01-21:
On archlinux aur found this problem: ponyprog: /lib exists in filesystem (owned by filesystem)
my patch:
```
--- a/distribution/posix/CMakeLists.txt	2020-01-21 08:29:49.949685608 +0000
+++ b/distribution/posix/CMakeLists.txt	2020-01-21 10:28:26.949800156 +0000
@@ -49,7 +49,7 @@
 # MESSAGE("PROJECT_SOURCE_DIR ${PROJECT_SOURCE_DIR}")
 
 INSTALL(FILES "${PROJECT_SOURCE_DIR}/udev/rules.d/90-ponyprog.rules"
-        DESTINATION "/lib/udev/rules.d/" 
+        DESTINATION "${CMAKE_SYSTEM_LIBRARY_PATH}/udev/rules.d/" 
         COMPONENT "application" 
         PERMISSIONS OWNER_WRITE OWNER_READ GROUP_READ WORLD_READ
         )

```

darkprof83, 2020-01-21:
```
darkprof@darkhost ~/aur/ponyprog $ LANG=en ls -al /
total 68
drwxr-xr-x  18 root root  4096 Jan 21 11:26 .
drwxr-xr-x  18 root root  4096 Jan 21 11:26 ..
-rw-------   1 root root  1024 May 16  2019 .random_number_seed
lrwxrwxrwx   1 root root     7 Nov 13 16:23 bin -> usr/bin
drwxr-xr-x   4 root root  4096 Jan 21 07:37 boot
drwxr-xr-x  21 root root  3900 Jan 21 11:09 dev
drwxr-xr-x 105 root root  4096 Jan 21 11:55 etc
drwxr-xr-x   7 root root  4096 Jan 14 07:52 home
lrwxrwxrwx   1 root root     7 Nov 13 16:23 lib -> usr/lib
drwxr-xr-x   2 root root  4096 Aug  7 17:25 lib32
lrwxrwxrwx   1 root root     7 Nov 13 16:23 lib64 -> usr/lib
drwx------   2 root root 16384 Dec 14  2018 lost+found
drwxr-xr-x   9 root root  4096 Jan 13 14:52 mnt
drwxr-xr-x   9 root root  4096 Jan 13 12:50 opt
dr-xr-xr-x 243 root root     0 Jan 21 11:08 proc
drwxr-x---  12 root root  4096 Jan 19 13:11 root
drwxr-xr-x  26 root root   760 Jan 21 11:09 run
lrwxrwxrwx   1 root root     7 Nov 13 16:23 sbin -> usr/bin
drwxr-xr-x   5 root root  4096 Dec 15  2018 srv
dr-xr-xr-x  13 root root     0 Jan 21 11:08 sys
drwxrwxrwt  22 root root   540 Jan 21 11:55 tmp
drwxr-xr-x  11 root root  4096 Jan 21 11:55 usr
drwxr-xr-x  13 root root  4096 Jan 21 11:09 var
```

lancos, 2020-01-21:
Debian/Ubuntu put rules in /lib/udev/rules.d (https://wiki.debian.org/udev), Arch Linux seems to use /usr/lib/udev/rules.d (https://wiki.archlinux.org/index.php/Udev)

lancos, 2020-01-22:
Checkout develop branch and test it with
cmake -DUSE_DEBUGGER=OFF -DUSE_QT5=ON -DUDEV_INSTALL_DIR="/usr/lib/udev/rules.d" ..
Tell me if it fixes your problem, please

closed 2020-02-05 by lancos