lancos/ponyprog #21: USB adapter support

GuilhermeMoreno, 2021-02-28:
According to PonyProg [website](http://www.lancos.com/prog.html) USB support have been on the TODO list for quite a while now. Issue #3 mentioned that it was the was first priority after porting it to QT. 

It seems that @eduard-x have done some progress on  reading an EEPROM using a CH341 based adapter, with plans to support CP2102 and PL2303 based adapters. Unfortunately, the branch seems to be inactive since 2019.

Is there any current plans to support USB adapters for EEPROMs? I saw that version 3.1.0 added support for PonyProgFT, but it seems to only mention AVR chips.

eduard-x, 2021-03-02:
There was no one to test the usb interface at the time, the driver of ch341 was decompiled with Ghidra  and i used different sources of source code like header files and other. We can implement it in the version 3.3. Currently is in developing the 3.2. I think, for this branch we have to many changes for testing. 
But i can program on if Claudio has nothing against a high-speed interface in his nice classical program.

lancos, 2021-03-02:
Hi,
at the moment the only USB interface supported by PronyProg is PonyProgFT (based on FTDI FT232H), it's a hi-speed solution and should work with all devices supported by PonyProg (not only AVR chips). What is lacking is the documentation, I should find the time to update the site.

GuilhermeMoreno, 2021-03-14:
> There was no one to test the usb interface at the time, the driver of ch341 was decompiled with Ghidra and i used different sources of source code like header files and other. We can implement it in [...]

I have a CP2102 interface with me right now but can get a CH341 based one if needed for testing. The code for the usb interface is in the [libusb-patches](/lancos/ponyprog/tree/wip/eduard/libusb-patches) branch, right?

It would be very useful to have that working!

> Hi,
> at the moment the only USB interface supported by PronyProg is PonyProgFT (based on FTDI FT232H), it's a hi-speed solution and should work with all devices supported by PonyProg (not only AVR chips). What is lacking is the documentation, I should find the time to update the site.

That's cool! If you could find some spare time to document the PonyProgFT interface it would be very helpful.

mrv96, 2022-11-07:
I'm intrested too about CH341 support