lancos/ponyprog #28: Build fails with LTO (odr violation)

eli-schwartz, 2024-03-13:
I tried to build with the following *FLAGS to optimize the build: `-flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing`

Note the -Werror=* flags are used to help detect cases where the compiler tries to optimize by assuming UB cannot exist in the source code -- if it does exist, ordinarily the code would be miscompiled, and this says to make the miscompilation a fatal error.

I got this error:
```
FAILED: ponyprog 
: && /usr/bin/x86_64-pc-linux-gnu-g++ -march=native -fstack-protector-all -O2 -pipe -fdiagnostics-color=always -frecord-gcc-switches -U_FORTIFY_SOURCE -D_FORTIFY_SOURCE=3 -fstack-clash-protection -flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing  -Wformat -Werror=format-security -std=c++11 -fno-exceptions -Wl,-O1 -Wl,--as-needed -flto=4 -Werror=odr -Werror=lto-type-mismatch -Werror=strict-aliasing -Wl,--defsym=__gentoo_check_ldflags__=0    -rdynamic CMakeFiles/ponyprog.dir/SrcPony/aboutmdlg.cpp.o CMakeFiles/ponyprog.dir/SrcPony/usbwatcher.cpp.o CMakeFiles/ponyprog.dir/SrcPony/main.cpp.o CMakeFiles/ponyprog.dir/SrcPony/Translator.cpp.o CMakeFiles/ponyprog.dir/SrcPony/e2app.cpp.o CMakeFiles/ponyprog.dir/SrcPony/at25xxx.cpp.o CMakeFiles/ponyprog.dir/SrcPony/at93cbus.cpp.o CMakeFiles/ponyprog.dir/SrcPony/crc.cpp.o CMakeFiles/ponyprog.dir/SrcPony/e24xx-1.cpp.o CMakeFiles/ponyprog.dir/SrcPony/e24xx-2.cpp.o CMakeFiles/ponyprog.dir/SrcPony/e24xx.cpp.o CMakeFiles/ponyprog.dir/SrcPony/e24xx-5.cpp.o CMakeFiles/ponyprog.dir/SrcPony/e2profil.cpp.o CMakeFiles/ponyprog.dir/SrcPony/fusemdlg.cpp.o CMakeFiles/ponyprog.dir/SrcPony/bitfield.cpp.o CMakeFiles/ponyprog.dir/SrcPony/blockmdlg.cpp.o CMakeFiles/ponyprog.dir/SrcPony/intfbuf.cpp.o CMakeFiles/ponyprog.dir/SrcPony/lpt_io_interf.cpp.o CMakeFiles/ponyprog.dir/SrcPony/pgminter.cpp.o CMakeFiles/ponyprog.dir/SrcPony/picbus.cpp.o CMakeFiles/ponyprog.dir/SrcPony/progoption.cpp.o CMakeFiles/ponyprog.dir/SrcPony/sernumdlg.cpp.o CMakeFiles/ponyprog.dir/SrcPony/x2444.cpp.o CMakeFiles/ponyprog.dir/SrcPony/at17xxx.cpp.o CMakeFiles/ponyprog.dir/SrcPony/at89sbus.cpp.o CMakeFiles/ponyprog.dir/SrcPony/at93cxx8.cpp.o CMakeFiles/ponyprog.dir/SrcPony/csmfbuf.cpp.o CMakeFiles/ponyprog.dir/SrcPony/e2cmdw.cpp.o CMakeFiles/ponyprog.dir/SrcPony/e2awinfo.cpp.o CMakeFiles/ponyprog.dir/SrcPony/easyi2c_interf.cpp.o CMakeFiles/ponyprog.dir/SrcPony/i2cbus.cpp.o CMakeFiles/ponyprog.dir/SrcPony/ispinterf.cpp.o CMakeFiles/ponyprog.dir/SrcPony/microbus.cpp.o CMakeFiles/ponyprog.dir/SrcPony/pic125xx.cpp.o CMakeFiles/ponyprog.dir/SrcPony/picbusnew.cpp.o CMakeFiles/ponyprog.dir/SrcPony/spi-bus.cpp.o CMakeFiles/ponyprog.dir/SrcPony/at250bus2.cpp.o CMakeFiles/ponyprog.dir/SrcPony/at89sxx.cpp.o CMakeFiles/ponyprog.dir/SrcPony/at93cxx.cpp.o CMakeFiles/ponyprog.dir/SrcPony/device.cpp.o CMakeFiles/ponyprog.dir/SrcPony/eeptypes.cpp.o CMakeFiles/ponyprog.dir/SrcPony/imbus.cpp.o CMakeFiles/ponyprog.dir/SrcPony/linuxsysfsint.cpp.o CMakeFiles/ponyprog.dir/SrcPony/pic12bus.cpp.o CMakeFiles/ponyprog.dir/SrcPony/ponyioint.cpp.o CMakeFiles/ponyprog.dir/SrcPony/rs232int.cpp.o CMakeFiles/ponyprog.dir/SrcPony/at250bus.cpp.o CMakeFiles/ponyprog.dir/SrcPony/at90sbus.cpp.o CMakeFiles/ponyprog.dir/SrcPony/binfbuf.cpp.o CMakeFiles/ponyprog.dir/SrcPony/dt006interf.cpp.o CMakeFiles/ponyprog.dir/SrcPony/e2dlg.cpp.o CMakeFiles/ponyprog.dir/SrcPony/filebuf.cpp.o CMakeFiles/ponyprog.dir/SrcPony/infomdlg.cpp.o CMakeFiles/ponyprog.dir/SrcPony/lpt_ext_interf.cpp.o CMakeFiles/ponyprog.dir/SrcPony/mpsse_interf.cpp.o CMakeFiles/ponyprog.dir/SrcPony/motsfbuf.cpp.o CMakeFiles/ponyprog.dir/SrcPony/pic168xx.cpp.o CMakeFiles/ponyprog.dir/SrcPony/portint.cpp.o CMakeFiles/ponyprog.dir/SrcPony/sde2506.cpp.o CMakeFiles/ponyprog.dir/SrcPony/wait.cpp.o CMakeFiles/ponyprog.dir/SrcPony/at250xx.cpp.o CMakeFiles/ponyprog.dir/SrcPony/at90sxx.cpp.o CMakeFiles/ponyprog.dir/SrcPony/busio.cpp.o CMakeFiles/ponyprog.dir/SrcPony/e2401.cpp.o CMakeFiles/ponyprog.dir/SrcPony/e2pfbuf.cpp.o CMakeFiles/ponyprog.dir/SrcPony/filldlg.cpp.o CMakeFiles/ponyprog.dir/SrcPony/interfconv.cpp.o CMakeFiles/ponyprog.dir/SrcPony/lptinterf.cpp.o CMakeFiles/ponyprog.dir/SrcPony/nvm3060.cpp.o CMakeFiles/ponyprog.dir/SrcPony/pic16xx.cpp.o CMakeFiles/ponyprog.dir/SrcPony/x2444bus.cpp.o CMakeFiles/ponyprog.dir/SrcPony/prefdialog.cpp.o CMakeFiles/ponyprog.dir/SrcPony/sdebus.cpp.o CMakeFiles/ponyprog.dir/SrcPony/moc_fusemdlg.cpp.o CMakeFiles/ponyprog.dir/SrcPony/moc_usbwatcher.cpp.o CMakeFiles/ponyprog.dir/SrcPony/moc_bitfield.cpp.o CMakeFiles/ponyprog.dir/SrcPony/moc_filldlg.cpp.o CMakeFiles/ponyprog.dir/SrcPony/moc_blockmdlg.cpp.o CMakeFiles/ponyprog.dir/SrcPony/moc_e2cmdw.cpp.o CMakeFiles/ponyprog.dir/SrcPony/moc_e2dlg.cpp.o CMakeFiles/ponyprog.dir/SrcPony/moc_aboutmdlg.cpp.o CMakeFiles/ponyprog.dir/SrcPony/moc_prefdialog.cpp.o CMakeFiles/ponyprog.dir/SrcPony/moc_progoption.cpp.o CMakeFiles/ponyprog.dir/SrcPony/moc_sernumdlg.cpp.o CMakeFiles/ponyprog.dir/SrcPony/moc_infomdlg.cpp.o CMakeFiles/ponyprog.dir/qrc_ponyprog.cpp.o -o ponyprog  /usr/lib64/libQt5Multimedia.so.5.15.12  /usr/lib64/libQt5PrintSupport.so.5.15.12  -lftdipp1  -lftdi1  -lusb-1.0  -lqhexedit  /usr/lib64/libQt5Network.so.5.15.12  /usr/lib64/libQt5Widgets.so.5.15.12  /usr/lib64/libQt5Gui.so.5.15.12  /usr/lib64/libQt5Core.so.5.15.12 && :
/var/tmp/portage/dev-embedded/ponyprog-3.1.3/work/ponyprog-3.1.3/SrcPony/pic168xx.cpp:67:3: error: type ‘struct IdType’ violates the C++ One Definition Rule [-Werror=odr]
   67 | } IdType;
      |   ^
/var/tmp/portage/dev-embedded/ponyprog-3.1.3/work/ponyprog-3.1.3/SrcPony/at90sxx.cpp:160:3: note: a different type is defined in another translation unit
  160 | } IdType;
      |   ^
/var/tmp/portage/dev-embedded/ponyprog-3.1.3/work/ponyprog-3.1.3/SrcPony/pic168xx.cpp:65:13: note: the first difference of corresponding definitions is field ‘code’
   65 |         int code;
      |             ^
/var/tmp/portage/dev-embedded/ponyprog-3.1.3/work/ponyprog-3.1.3/SrcPony/at90sxx.cpp:157:13: note: a field with different name is defined in another translation unit
  157 |         int code1;
      |             ^
lto1: some warnings being treated as errors
lto-wrapper: fatal error: /usr/bin/x86_64-pc-linux-gnu-g++ returned 1 exit status
compilation terminated.
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```
Originally reported downstream: https://bugs.gentoo.org/855272
Full build log: [build.log](https://github.com/lancos/ponyprog/files/14592026/build.log)

lancos, 2024-03-23:
Fixed

closed 2024-03-23 by lancos

eli-schwartz, 2024-03-24:
Backported the commit locally and verified it fixed my issue. Thanks!